JosephHewitt/wardriver_rev3 #4: Required ESP32 board version is 2.0.0

lozaning, 2022-05-07:
`Please install version 2.0.0 (at the time of writing, later versions have not been tested but might work).`

Just confirming that you definitely want to be on the 2.0.0 version of the esp32 board version. I was on the 2.1.3 version for another project and the B board code was failing to compile due to some onewire issues. Downgrading to the 2.0.0 version and trying again worked fine.

JosephHewitt, 2022-05-07:
I have just reproduced this on ESP32 board version 2.0.3 with OneWire 2.3.6.

Version 2.0.0, 2.0.1, and 2.0.2 are unaffected by this issue.

Since this is an issue with compiling the OneWire library, I suspect it may be fixed in a later ESP32/OneWire update.

For now, be sure to use ESP32 version 2.0.0 - 2.0.2 until a solution to the problem is found.

JosephHewitt, 2022-11-20:
I can no longer reproduce this issue on the latest versions (ESP32 2.0.5 and OneWire 2.3.7). This issue appears to be resolved.

I will update the documentation as part of the Wardriver 1.0.1 release.

closed 2022-11-20 by JosephHewitt