JosephHewitt/wardriver_rev3 #9: SD card not found.

grymoire, 2022-07-17:
I have a 32GB newly formatted Samsung SD card (using the Microsoft formatter) and a  [Micro SD Storage Expansion Board Mini Micro-SD TF Card Memory Shield Module With Pins for Arduino ARM AVR](https://www.aliexpress.com/item/2251832681257883.html ). But no SD card is found.  - SD.begin() fails.I'm trying to get further into the debugging but I wondered if you can give me a few tips if you have a;

JosephHewitt, 2022-07-17:
Hi,

Please ensure that your card is formatted to FAT32 and not exfat or FAT16. If your card is definitely formatted correctly, you may have found an issue I have spotted once before:

I have noticed an issue with a Sandisk Ultra 32GB card and thought the issue was limited to that particular brand but it seems that certain 32GB cards are doing something which the Arduino ecosystem doesn't support. I have not yet seen this issue happen with a card under 32GB.

I'm still trying to determine exactly how widespread this issue is and what the exact cause is.

Right now, my best guess is that some micro SD cards have bad SPI implementations and this issue is referenced on the Flipper Zero project blog: [see here](https://blog.flipperzero.one/quick-start-guide-and-shipping-update/#what-sd-card-to-get). Specifically some cards may work perfectly in a PC using SDIO but then fail to work correctly on a slower device (such as the Wardriver) which uses SPI.

The Wardriver writes a very small amount of data so my recommendation is to try a 16GB card which should still offer plenty of storage capacity.

Please let me know if you have issues with other cards. I will add a warning to the wiki regarding 32GB SD cards.

grymoire, 2022-07-18:
Thanks for the reply. I'll get some smaller  cards and try again.

grymoire, 2022-07-23:
I found the problem - bad solder joint! Thanks. It's working now.

grymoire, 2022-07-28:
Fixed

closed 2022-07-28 by grymoire