JosephHewitt/wardriver_rev3 #11: Special characters not parsed in WiFi configuration

JosephHewitt, 2022-07-19:
When configuring a WiFi network to connect to or as a fallback network, special characters can cause issues with the parsing.

Specifically, the HTML form uses GET to transfer the values to the wardriver and characters such as `?` and `&` are mishandled.

Switching to POST is probably the best solution which is more suited for the task anyway.

K1rdro, 2022-10-25:
My workaround for this issue was to print the password on the screen when the device boots. I found it added a couple % signs to mine.

JosephHewitt, 2022-11-20:
Fixed with PR #22

It would still be beneficial to update this code to use POST requests in the future but for now that is not a priority.

closed 2022-11-20 by JosephHewitt