JosephHewitt/wardriver_rev3 #13: Switch/button for changing between captive portal and scanning mode

jpdias, 2022-07-28:
Add some mechanism (e.g. button, switch) for switching between captive portal and scanning mode without requiring to reboot the device. This also allows for the device to automatically enter in scanning mode on boot, and the user can change to captive when needed (e.g. with a two-state switch).

JosephHewitt, 2022-07-28:
This is a good idea and something I had considered before but it does have an unfortunate downside: With the current PCBs, adding a switch won't be a very user-friendly thing to do meaning I expect most users won't do it. For that reason, this feature would have to be off by default and then enabled if needed.

My primary focus for now is to add features which will work on all wardrivers (eg, the "vanilla" devices) so this isn't something I will implement any time soon. 

I'll leave this issue open and maybe someone else will implement it; I'll merge it in if I get a PR which doesn't affect vanilla devices.

jpdias, 2022-07-29:
@JosephHewitt I totally concur with that approach, having something in the initial configuration screen to set the operation mode with or without the switch. If I got the time I will try to look into this enhancement in the terms presented.