JosephHewitt/wardriver_rev3 #18: WPA and WPA2 Personal network shown as [OPEN]

grymoire, 2022-08-05:
I have a local network with WPA & WPA2 Personal security.
I've setup my wardriver system to join this network.
When it collects information, it always lists the SSID as [OPEN]

grymoire, 2022-08-05:
To be precise, the router has these options enabled

WPA2 Personal |  
WPA2 Personal with SHA256 
WPA3 Personal / SAE

I will try different settings.

JosephHewitt, 2022-08-05:
Thank you.

I suspect this issue could be related to WPA3 being enabled since I haven't seen anything like this before when testing WPA/WPA2 only networks.

I'll try to set up a test WPA3 network at some point to see if I can reproduce the issue.

JosephHewitt, 2023-04-27:
I have also recently observed this issue. Some networks are written to the Wigle CSV file with no information for the security type which Wigle then seems to interpret as an open network.

I'm still unable to create a WiFi network which triggers this issue, but now I have some more information which should help with the troubleshooting.

[This espressif page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv416wifi_auth_mode_t) lists the security types, and there are some entries which aren't included in my `security_int_to_string` function, which is likely to be the culprit.

I will keep investigating this issue.

JosephHewitt, 2023-05-01:
The `security_int_to_string` function was missing a `break` in the switch case causing WPA3/WPA2 PSK networks to return a blank string for the security type which Wigle interprets as an open network.

This has been addressed in #54 which is now in `main`. A beta version will be released shortly which includes this fix.

I will keep this issue open for now so I have some time to test if the issue is still present.

JosephHewitt, 2023-05-28:
This now appears to be resolved so I will close this issue. Please feel free to open another issue if you see similar problems on the latest versions.

Thanks for reporting this and I am sorry it went so long before being resolved.

closed 2023-05-28 by JosephHewitt