JosephHewitt/wardriver_rev3 #20: wifi not connecting

clint70, 2022-09-04:
First off, thank you for the great project!!  I love this, I have it assembled and working, well sorta.  As of right now, the display shows wifi XX, ble XX etc... but there is not test.txt file and on start up all I get is connecting.... / failed to connect.  Time does not sync etc.

I have tried multiple AP, with differing names (ie. no special char, no spaces etc) and still no success.  I am not sure where to start with the trouble shooting.  It is strange, as the first time setup does show the wifi AP etc.... I enter the psk and seems ok.  BUT on boot, no connection??  Is there a log I can check?  OR ... some suggestion?

Thanks for the time / project, it will be fun once up and running... ;)

clint70, 2022-09-05:
still can’t get connection… but debugged the ssid/psk for some reason there are added char. in this case a ‘+’

clint70, 2022-09-05:
Ok... so from some minor debugging, I have found that the current code DOES not like spaces in the SSID, it adds '+' as mentioned above.  With a small change to the code, 

```String con_ssid = preferences.getString("ssid","");
  String con_ssid_fixed = con_ssid.c_str();
  con_ssid_fixed.replace("+", " ");
  String con_psk = preferences.getString("psk","");

This obviously replaces the '+' sign. 

Now I have run into once connected to my 'home net', there is no HTTP page, it just spins?  I also noticed that the ping time for the dev. is very high

<img width="1862" alt="image" src="https://user-images.githubusercontent.com/8769072/188491046-19c9cb3c-592c-4c92-9ef9-9c296e35c265.png">

JosephHewitt, 2022-09-05:
Hi,

The issue regarding the `+` is already reported on issue #11. While replacing the `+` with a space is a reasonable quick fix, I will be switching to POST requests for the WiFi config in the future which will fix all URL encoding issues, such as this one.

The high latency and blank page appears to be an unrelated issue. Just to confirm, do your ESP32 boards have antennas connected? A weak WiFi connection can cause this kind of problem.

clint70, 2022-09-05:
Thank you for the reply... and I did read the issue regarding the URL encoding issue, was not sure that was the problem.  BUT understood.  I am glad it connects now, 

The high latency and blank page, issue? I have not figured out yet.  AND to answer your question, the antenna are connected AND the device is close to the AP.  So there should be no latency... I am assuming here, that the ESPs are slow to reply? OR doing something with the connection for the webpage... BUT I am not sure.  

Question, is there a place to look for log files?  OR something like that?  Newer to arduino debugging

JosephHewitt, 2022-09-06:
Hello,

There isn't much debugging enabled in my software for the web interface. You should see "client connected" on the LCD when your browser successfully connects to the wardriver but there isn't much else to indicate whether the HTTP communication is/isn't working.

Could you try to connect to your wardriver using `curl` on your computer? This will rule out any issues with your browser. The exact command for you should be `curl http://10.10.20.120/`

Alternatively, could you attempt to connect from another device such as a smartphone?

clint70, 2022-09-06:
Ok... I'll see what i can figure out. Yes i have tried different devices, browsers and no success.  

Not sure best way to proceed, but I'll dig into it a bit.

clint70, 2022-09-06:
So it connects, displays 'client connected' on the lcd.  I can ping the host, but when I curl / web browse to the IP, no reply??  Any ideas to get me going on troubleshooting?

[image]

clint70, 2022-09-08:
So it connects, displays 'client connected' on the lcd.  I can ping the host, but when I curl / web browse to the IP, no reply??  Any ideas to get me going on troubleshooting?

JosephHewitt, 2023-04-05:
Hi,

I'm sorry this issue went without a response for so long.

Unfortunately this is a very hard issue to diagnose without some extensive debugging on your end, and I'm not sure how we would go about debugging this particular issue.

The `client connected` message on the LCD means that a HTTP connection was initiated. This shows the ESP is connected and listening correctly for connections so I'm not sure why it would then not return any data.

Did you ever find the cause of this? It's something I've never been able to reproduce.

closed 2023-05-11 by JosephHewitt