JosephHewitt/wardriver_rev3 #24: What library gives parseUtils?

barry99705, 2023-03-22:
C:\Users\Barry\Desktop\wardriver_rev3-1.0.1rc1\wardriver_rev3-1.0.1rc1\A\A.ino:1:10: fatal error: parseUtils.h: No such file or directory
 #include <parseUtils.h>
          ^~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: parseUtils.h: No such file or directory

JosephHewitt, 2023-03-22:
I'm not actually sure, I need to research this some more.

In the meantime, ensure you have the correct board selected. Sometimes having something else (such as an Arduino or the ESP8266) can lead to this kind of issue.

JosephHewitt, 2023-04-02:
This was added in PR #22

The [GParser](https://github.com/GyverLibs/GParser) library is now required.

It does appear that parseUtils is unused though, so removing that line may work for you if it still causes you issues.

barry99705, 2023-04-02:
Wiped out all my arduino stuff and redownloaded, works now!

closed 2023-04-02 by barry99705