JosephHewitt/wardriver_rev3 #31: Dropping connection to network

bennybo, 2023-04-06:
I recently completed the hackerbox 089 project and it is now working fine. I can remove the sd card and read captured data with no problems. I can also attach to the wardriver.uk SSID and setup various networks I have in the house and after powering up the display shows the IP it received from DHCP but will drop the connection after about 8 seconds. I checked power, different USB cables, different networks for main and backup with the same results. During the short time it connects I cannot bring up a web browser with the displayed IP either.

Brian.

JosephHewitt, 2023-04-06:
This potentially sounds like the same issue as #20 which currently has no known steps to reproduce or fix.

You state that the connection is dropped after about 8 seconds, what are the exact symptoms of this? eg, does the LCD state something different?

Just to confirm, are you using version 1.0.1 from the `Releases` tab on the side of the repo? This version should be used as it is the latest stable version.

Which version of the ESP32 board software do you have installed? This can be checked in `Tools` -> `Board: xx` -> `Boards Manager` and then search for `ESP32`.

Did you ensure every option in the `Tools` menu was configured as stated in the readme? Such as:

- Board: `ESP32 Dev Module`
- Upload Speed: `921600`
- CPU Frequency: `240MHz (WiFi/BT)`
- Flash Frequency: `80MHz`
- Flash Mode: `DIO`
- Flash Size: `4MB (32Mb)`
- Partition Scheme: `Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)`
- PSRAM: `Disabled`
- Arduino Runs On: `Core 1`
- Events Run On: `Core 1`

If everything is configured correctly and you are still getting the same issue, please could you connect the ESP "Side A" to the Arduino IDE serial monitor and then reset it? You should see a lot of messages printed during boot, maybe some of these will help track down the issue. Ideally please enable the `show timestamp` in the bottom left of the serial monitor so it is clearer how long each step is taking. Please share this whole output here and censor any personal information, if any.

Right now, I don't really have any suggestions on how to fix the issue since it is something I have never been able to reproduce.

bennybo, 2023-04-06:
Yes, it is 1.0.1 release, The ESP32 board software is 2.0.7. I double checked the options in tools and they are correct.

Here is the output from the serial monitor before it starts collecting data.

18:37:22.873 -> Attempting to connect to WiFi.
18:37:23.017 -> Connected, getting the time
18:37:27.672 -> Got time from NTP: 1680820648
18:37:27.672 -> Time is now set to 1680820648
18:37:27.672 -> Continuing..
18:37:27.672 -> 192.168.1.130
18:37:37.696 -> Disconnecting
18:37:37.977 -> Opening destination file for writing
18:37:37.977 -> /wd3-10.csv
18:37:38.008 -> Added 10 seconds to epoch

I monitor connected clients on my routers admin page and I can watch it hand out the IP and then of course it disappears upon disconnecting.

Brian.

JosephHewitt, 2023-04-07:
Hi,

This is normal behaviour but I admit 10 seconds is too short. As part of #32 I will allow this timeout to be adjusted. The idea is the wardriver connects to your network, gets the time, and then starts the normal wardriving mode. This timeout used to be larger but was lowered during early testing/development and then never raised again. If a web browser connects to the interface during this short time period, the timeout is raised giving you time to interact with it.

The timeout should be around 1 minute if the wardriver is broadcasting a network rather than connecting to one, giving you more time to connect a device.

You also state that "During the short time it connects I cannot bring up a web browser with the displayed IP either" which is something I can't reproduce. It might be worth manually adjusting the timeout to see if it still happens when you have plenty of time to connect. You can do this by editing the following line (around line 346) :
`unsigned long disconnectat = millis() + 10000;` to
`unsigned long disconnectat = millis() + 60000;`

This will cause the device to connect to your network for a minute before continuing into the normal wardriving mode. I can investigate some more if you are still unable to open the web interface after this change.

bennybo, 2023-04-08:
Ok. That did the trick. Everything is working great now. Thanks for your time on this.

Brian.

JosephHewitt, 2023-04-08:
Hi,

Thanks for your feedback regarding this. I hope to make this more clear for future users of the project by improving the documentation (#34) and by adding some feedback to the LCD during boot (#33).

closed 2023-04-08 by JosephHewitt