JosephHewitt/wardriver_rev3 #50: Show local time on the LCD

JosephHewitt, 2023-04-27:
It would be useful to display the local time on the LCD instead of UTC.

The quick/easy option is to retrieve the timezone with Javascript whenever the user accesses the interface, but this means it will not update automatically when the user moves.

The better solution is to use something like [ZoneDetect](https://github.com/BertoldVdb/ZoneDetect) which converts coordinates to a TZ, but this adds quite some overhead which is probably overkill for such a small feature.

pejacoby, 2023-04-30:
This is a great idea -- I'd be OK with a simple /etc/TZ style file in the file system that could be set to the appropriate time zone.

anyconhhv, 2024-01-23:
It would also be great to have an accurate timestamp on the files on the SD card.