JosephHewitt/wardriver_rev3 #51: Download files naming format and extension

pejacoby, 2023-04-30:
When downloading files from the web interface (1.1.0b1) they are saved with the double-extension ".csv.html".
Could that be just ".csv" instead?  Minor issue but I end up renaming them all the time.

It would also be nice to have the filename be a timestamp, another element I end up adding to all files.  YYYY-MM-DD would be plenty to add, as the file number keeps things unique.  YYYY-MM-DD-HH-MM would be another option.

JosephHewitt, 2023-04-30:
Thanks for opening this issue.

I've found the most likely culprit for the `.html` extension being added. The filename is set in the `Content-Disposition` HTTP header, however a somewhat conflicting `Content-type: text/html` is sent on all requests. Many browsers will use the unaltered filename as sent by the header, but some browsers will add `.html` since it has been told a HTML file is being sent. Most likely changing this to `text/csv` for the file download will fix the issue.

As for the date being added to the filename, I like the idea and it should be quite easy to implement.

I will work on this soon.

JosephHewitt, 2023-05-01:
The correct `Content-Type` header is now sent with #52

The date is now included with the filename of a downloaded file with #53 

These changes are now in `main` but a new beta version will be released soon with these changes. Thanks for opening this issue.