JosephHewitt/wardriver_rev3 #56: Close <HTML> tag at page ends

pejacoby, 2023-05-08:
I was looking at why the time between the WiSpy showing it had 'got one' for an HTTP request to the server and a response on the client is so long.  It's minor, but the pages aren't closing the <HTML> tag with a corresponding </HTML> tag.  This might(?) be causing Safari to do a big-think expecting more input.  Closing the pages would be be good form and format in any case.

JosephHewitt, 2023-05-09:
Hi,

Thanks for letting me know about this, the missing tags were not intentional so I will add them soon.

If it is only the main page (with the list of files to download) that is slow, you are probably seeing the affects of issue #2 where it loads slower as more files are saved.

pejacoby, 2023-05-09:
I should have been more specific- the slowness is between the time I tap on a file download link and when the download prompt appears on my phone. This might just be the time it takes the reader to open the file for sending. I’ll do some more timing to see if file size makes a significant difference- it doesn’t seem to.

pejacoby, 2023-05-11:
I did some further testing and the slowness appears to be a SAFARI browser issue.  Both on iOS and MacOS.  Regardless of the number of files on the SD card -- 30 or just 1.

If I use Firefox on MacOS to download files, results are instantaneous.

If I use Safari on iOS, there is a consistent 28-30 second delay after tapping a download link and getting the "Save" prompt to save the file.
If I use Safari on MacOS, after clicking the download link the browser will stall and eventually the WiSpy transitions into wardrive mode and the file never downloads. During the wait, Safari shows _no_ activity in the Network trace window.

In either Safari case, while stalled the WiSpy shows "Client connected.  Awaiting request..."

I'll see what else I can find, but definitely seems to be a Safari thing.

pejacoby, 2023-05-13:
Thanks for checking.  I had not tried cancelling and retrying the download -- that does work every time as you noted.  I'm curious to see if the HTML tag changes have any affect on Safari.
I realize I'm bleeding into another issue here, I can open a separate item if that makes more sense.

JosephHewitt, 2023-05-20:
I looked more into this issue today. Whenever this issue occurs, Safari is waiting for a response from the wardriver which never arrives. From the perspective of the wardriver, Safari never actually sends a request, it simply opens the TCP connection and leaves it idle; this means the wardriver never sends a reply and the connection stays idle from both sides until timeout. 

I have implemented a short timeout in #58 which closes the TCP connection after 750ms if no data is sent by the client. This prompts Safari to re-open the connection and try again. I have only tested this on iOS and it seems to work well.

While I did not find the root cause of this, the problem now seems to be resolved. Since the issue was happening at a lower level than HTTP, the closing HTML tags made no change. If I ever refactor the HTML, I will properly open/close all relevant tags, but for now I will leave the HTML as-is since that part of the code is messy and hard to maintain.

I will make a new beta release shortly which will include this change. I will also leave the issue open for a while in case there are any issues with this fix.

Thank you for opening this issue.

pejacoby, 2023-05-25:
I have loaded 1.1.0rc1 and it is working very well with Safari on iOS and MacOS 12.6.6.  Thanks for the fix!

JosephHewitt, 2023-05-28:
Great, thank you for confirming.

closed 2023-05-28 by JosephHewitt