JosephHewitt/wardriver_rev3 #71: Implement OTA updates

JosephHewitt, 2023-06-04:
Now that binaries are generated automatically, over the air updates are much easier to implement.

It would be best to deliver the binary over HTTPS and use custom certificates (eg, self signed) so that rotation of certificates can be in our control. This would be implemented by having a primary and a standby certificate in the firmware. The server would use the primary until rotation is needed (such as the cert expiring) then the standby would take the place of the primary, a new cert would be generated, and then the new cert would be pushed in a firmware update to replace the expired primary.

This requires us to host the binaries outside of GitHub (in addition to GitHub) so we can have a server with the custom HTTPS certs. This is easy, but it would be nice to have some kind of proof the same binaries are being hosted so people can see that we're not trying to sneak hidden stuff onto the wardrivers.

It would also be nice to have an upload form on the ESP web interface so people can generate (or download) the binaries themselves, do whatever checks they deem necessary, and then upload once they are satisfied.

We also need to find a nice method to update both ESPs. The obvious solution is to allow ESP B to connect to WiFi and run the same OTA code as A. This does mean we need to transfer WiFi credentials from A->B which could introduce a low complexity attack against the device. The firmware binary could be sent over serial from A, but we need to verify integrity of the transfer fully.

pejacoby, 2023-06-04:
In the pipeline can you generate a SHA256 hash of the build.zip (or it's contents) that the user (or the OTA process) can check after downloading to validate the code is unaltered?

JosephHewitt, 2023-06-05:
I agree with generating hashes for the binaries, this also solves the problem with flashing B. The binary can be sent over UART from A along with the hash which B can verify. This helps ensure the file was not damaged during the transfer over UART.

Although it's not strictly necessary for security, I think it's important for the user to be able to verify the binary they are installing is identical to the one generated by the pipeline. This can be achieved by downloading the binary on the wardriver, calculating the hash, and displaying it somewhere before the update is installed. The user can manually check the hash before approving the update.

The updates would be served from my infrastructure using the HTTPS method mentioned in the original post; this saves the effort of maintaining a CA trust store on the device which would offer us no benefit anyway.

JosephHewitt, 2023-07-08:
Completed in #83, released in 1.2.0b1.

closed 2023-07-08 by JosephHewitt