JosephHewitt/wardriver_rev3 #81: Side A stopped Booting

MadManX99, 2023-06-21:
I'm pretty new to these boards (and microcontrollers in general)... I'm using a hardware version from Hacker Boxes and the issue started with Side A only (v1.1.0rc).  I noticed the screen was black and tried power cycling, no luck.  I also checked the SD Card and no data was written to it.  I've been using the project for several weeks now (over 171k new additions to Wigle since getting this working).

Originally using Firmware v1.1.0rc. When connecting to a PC, Side B works fine and I can see the serial monitor feeding out scanned data.  If I take the exact same cable and plug it into side A... The COM port just cycles connected/disconnected. Putting it into Download mode... I was able to upload a basic sketch to confirm the board isn't dead.

void setup() {
  Serial.begin(115200);
  Serial.println("Starting");
}
void loop() {
  Serial.println("Running...");
}

But, if I upload Side A (v1.1.0) or Side B to the same board, it goes back to cycling the COM port. I now have the board sitting on the bench with nothing attached to it.  So I'm not sure if its software crashing something, or the board died.

test.txt, the last few entries were since updating to 1.1.0rc:
_BOOT_1.1.0rc1, ut=2266413, rr=1, id=4003788, bc=86, ep=1685629906
_BOOT_1.1.0rc1, ut=2443699, rr=1, id=4003788, bc=87, ep=0
_BOOT_1.1.0rc1, ut=2377789, rr=1, id=4003788, bc=88, ep=1685630756
_BOOT_1.1.0rc1, ut=2269792, rr=1, id=4003788, bc=89, ep=1685630974
_BOOT_1.1.0rc1, ut=2394085, rr=1, id=4003788, bc=90, ep=1685637787
_BOOT_1.1.0rc1, ut=2427782, rr=1, id=4003788, bc=91, ep=1685661791
_BOOT_1.1.0rc1, ut=2364740, rr=1, id=4003788, bc=92, ep=1685740600
_BOOT_1.1.0rc1, ut=2356746, rr=1, id=4003788, bc=93, ep=0
_BOOT_1.1.0rc1, ut=2393182, rr=1, id=4003788, bc=94, ep=0
_BOOT_1.1.0rc1, ut=2437381, rr=1, id=4003788, bc=95, ep=0
_BOOT_1.1.0rc1, ut=2434632, rr=1, id=4003788, bc=96, ep=0
_BOOT_1.1.0rc1, ut=2421691, rr=1, id=4003788, bc=97, ep=0
_BOOT_1.1.0rc1, ut=2389497, rr=1, id=4003788, bc=98, ep=0
_BOOT_1.1.0rc1, ut=2372779, rr=1, id=4003788, bc=99, ep=0
_BOOT_1.1.0rc1, ut=2432408, rr=1, id=4003788, bc=100, ep=1686098241
_BOOT_1.1.0rc1, ut=2352215, rr=1, id=4003788, bc=101, ep=1686185254
_BOOT_1.1.0rc1, ut=2436653, rr=1, id=4003788, bc=102, ep=1686270357
_BOOT_1.1.0rc1, ut=2419991, rr=1, id=4003788, bc=103, ep=1686337353
_BOOT_1.1.0rc1, ut=2438708, rr=1, id=4003788, bc=104, ep=0
_BOOT_1.1.0rc1, ut=2414774, rr=1, id=4003788, bc=105, ep=0
_BOOT_1.1.0rc1, ut=2430203, rr=1, id=4003788, bc=106, ep=0
_BOOT_1.1.0rc1, ut=2393580, rr=1, id=4003788
_BOOT_1.1.0rc1, ut=2405253, rr=1, id=4003788, bc=108, ep=0
_BOOT_1.1.0rc1, ut=2383849, rr=1, id=4003788, bc=109, ep=1686704961
_BOOT_1.1.0rc1, ut=6397698, rr=4, id=4003788, bc=110, ep=1686705037
_BOOT_1.1.0rc1, ut=2396251, rr=1, id=4003788, bc=111, ep=1686791300
_BOOT_1.1.0rc1, ut=2385380, rr=1, id=4003788, bc=112, ep=1686876561
_BOOT_1.1.0rc1, ut=2432352, rr=1, id=4003788, bc=113, ep=0
_BOOT_1.1.0rc1, ut=2403651, rr=1, id=4003788, bc=114, ep=0
_BOOT_1.1.0rc1, ut=2460080, rr=1, id=4003788
_BOOT_1.1.0rc1, ut=6447069, rr=7, id=4003788

When updated to 1.1.0, no log is ever created.  I also don't see the screen still or the GPS powering up (blinking light) and cycling the COM port.

Having a USB power meter connected, seems to never pull more that 0.04A with just the board.  The entire project (both boards, gos, gsm, sd card, and with an added small fan) pulls 0.35A

JosephHewitt, 2023-06-21:
Hi,

Thanks for the report. I have seen this kind of behaviour with the ESP boards during over current conditions or when the voltage regulator is faulty. It may also mean your power supply is unable to deliver enough power; some PCs struggle with this so I would also test with a power bank and/or phone charger along with a different cable just to rule that out.

You state that you added a fan, how is this attached and what voltage is it? It is possible this has caused some damage depending on how it is connected. Is it possible to temporarily disconnect it? If so, I'd try that to see if the device is able to boot normally again. 

When you uploaded your test sketch, did it work flawlessly even when unplugging and replugging to the PC or did that also have reliability issues?

Good luck and let me know how it goes.

MadManX99, 2023-06-22:
Thanks for the feedback...

Power Supplies have been just one PC that I originally used to set everything up and test on, I haven't tested the max current of the USB port, but I haven't seen the project pull more than 500mA yet.  I've also used a 100w capable battery (actually the one it was connected to when it stopped) and a 15w capable battery one as well.  Both are quality brands and I've used them to run tablets, Pis, and even a laptop.  So I'm hoping it wasn't that.

The fan is an "Easycargo" Pi-Fan Model EDL3007305 DC 5v @ 0.12A (30mm x 30mm).  Its connected to the Ground and 5V pins for power with nothing  isolating it.  With two boards, the GSM board, GPS, screen, and fan, it usually pulls around 0.25A (my power banks display bounces between 0.2-0.3A.  I never saw it over 0.3A on my bench power supply when initially building it.  I do have a USP Port Meter and a Bench Top power supply.

I live in Arizona, and have been carrying it in a pack, so I've added small heat sinks to the ESP32 chips on each board and the fan blows directly on them (it barely moves air, but its more than nothing).  Its easily over 105F outside now in the shade.  I've noticed it would stop updating the screen occasionally and unplugging it from the battery for a few minutes and plugging back in resolved it (just cycling it didn't always work, usually locked up again pretty soon after).  That's why I added the sinks and small fan to help keep it closer to ambient since that seemed to be well within the chips spec sheet.  In the same pack I'm also running a headless Pwnagotchi (Pi Zero W v1) with nothing more than the board and it's own battery, or shared battery on the one with multiple ports.

As for testing now... I had the board on header pins, so its been totally removed (sitting my my desk with only the USB cord to the PC) from the project and will only load the firmware if put into download mode manually.  A simple sketch like above, after loading boots and outputs to the IDE, but reloading Side A or B code now immediately starts cycling the COM port. I wasn't sure what other sketches I should try to mess with to see if they ran or not.

I don't have a temp sensor in this project (looking into how/where I can add that near my PCB), so I'm not sure the overall temps.  The Pwnagotchi has reported up too 58C I think I saw once.  It has a tight enclosure with a heat sink as well, But I've been keeping the top off it's enclosure, so I'd guess my bag as a whole could be getting up that high on these boards as well.

I've got 3 more of these boards due today... Crossing my fingers that I can either save this one, or not toast another.  If you think it might be the fan, would possibly adding a diode to the + or - help isolate it from the board?  I'm taking a wild guess on this fan that it could be brushless, but I don't think it would create any current if held in place while running.  Or any other better suggestion, since I just grabbed something meant for a Pi and fit my size and power requirements.  I'm also looking to add a UPS battery (rated at 3A peak) to the project as well, but trying to find a charging port for it to power the entire project (ideally USB-C port for ease).  The HackerBox going me going, and I've enjoyed  your project and supporting Wigle, so Looking forward enhancing my version, and correcting any hardware issues I may create... Of if they are software related, reporting them with as much info as I can (I'm also a SQL database/dot net developer).

JosephHewitt, 2023-06-25:
I suspect this isn't an issue with the code since this doesn't seem to be a widespread issue. I do find it odd that the problem started when you went from version 1.1.0rc to 1.1.0 though because those versions are identical but with an updated version number. 

Just to confirm, when flashing the wardriver firmware, did you set the `Flash Mode` to `DIO`? I've seen some unusual behavior when other modes are used on some boards.

It's possible that the inrush current when you first turn on the device is causing a voltage drop which resets the ESP32, but I don't think the symptoms match exactly for that to be the case. The fan in addition to all the other components starting up might be just a little too much. Keep in mind that although you are using a powerful battery/supply, it probably won't actually be able to supply the full amount of current since it has to maintain the USB spec. I haven't checked, but I suspect most of these ESP32 boards are USB2 meaning 5v/0.5A. I have seen quick spikes of up to around 0.4A in my testing, so a 0.12A fan could potentially cause that limit to be exceeded for brief periods.

When running other code (such as repeating a string constantly over serial) the current will be slightly less since the WiFi radio likely won't power up, the screen will stay blank, etc. 

Adding extra components to the fan may help but you can probably determine if the fan is the issue by temporarily removing it first. A diode and a capacitor before the fan will probably reduce the chance of power related issues, but it still doesn't explain why your build worked correctly for a while before stopping.

JosephHewitt, 2023-07-16:
Hi,

Were you able to make any progress with this issue?

closed 2023-07-22 by JosephHewitt