JosephHewitt/wardriver_rev3 #103: Support larger LCDs

JosephHewitt, 2023-08-13:
Right now the LCD parameters are hardcoded. It would be good to allow larger displays with larger font sizes.

Shadesh, 2023-08-13:
Just make the variables

SCREEN_WIDTH 128
SCREEN_HEIGHT 32 

adjustable via the cfg.txt

JosephHewitt, 2023-08-13:
That is the general idea, however the display is initialized much sooner than the `cfg.txt` being loaded. The display is meant to be one of the first things to load so that any early errors can be displayed on boot.

Hopefully if a larger display is used, it can still be initialized using the old values then can be re-initialized using the correct values later on. This would mean that any boot errors are displayed in a small font, but that's probably a "good enough" implementation.