JosephHewitt/wardriver_rev3 #107: LCD Not Displaying Properly

K0R41, 2023-08-25:
[image]

I've just assembled the rig, but the LCD is not displaying properly (Image Attached)

I have the BW16 installed instead of the GSM.  I Flashed the ESPs with the files from the main branch.

Any idea what may be causing this or how I can fix it?

Thanks

JosephHewitt, 2023-08-25:
I suspect this is either a fault with the LCD controller or bad wiring/soldering between the screen and PCB.

I would suggest replacing the screen if the wiring and soldering looks okay.

K0R41, 2023-08-26:
I have re-soldered the same LCD with the same result.
I then tried a different LCD and also have the same result.

JosephHewitt, 2023-08-26:
This is not something I have seen before. To rule out software issues, you could try installing an older version of the software from binary; this will rule out any issues which could have been caused by the software complilation on your PC.

You can do that by following the steps on the main README under the heading "Flashing (from binary) [BETA]". You can find the binaries on the Releases tab in the "Assets" dropdown of each release. You need the `build.zip` file.

K0R41, 2023-08-27:
Yes it seems to have been a hardware/soldering issue somewhere along the line.
I have remade it on a 2nd PCB with no issues.
Thank you

JosephHewitt, 2023-08-27:
Hi,

Thanks for the update. Please let me know if you have any other issues.

closed 2023-08-27 by JosephHewitt