JosephHewitt/wardriver_rev3 #110: Unable to compile A board - SET_LOOP_TASK_STACK_SIZE(10240)

gjchester, 2023-08-29:
Reflashing prior to adding a BW_16 module and while B flashes without error, A will not compile.  

Arduino software is 1.8.19   All the requirements on the Wardriver page are met.  Code was pulled from the latest Github branch on the 28th Aug2023.

The error seen is 

> 
> Using library Adafruit_BusIO at version 1.14.1 in folder: C:\Users\GJC\OneDrive\Documents\Arduino\libraries\Adafruit_BusIO 
> Using library Adafruit_SSD1306 at version 2.5.7 in folder: C:\Users\GJC\OneDrive\Documents\Arduino\libraries\Adafruit_SSD1306 
> exit status 1
> expected constructor, destructor, or type conversion before '(' token
> 

and the Arduino software highlights line 28 as being the issue.  

```
//The stack size is insufficient for the OTA hashing calls. This is 10K, instead of the default 8K.
SET_LOOP_TASK_STACK_SIZE(10240);

```

Commenting out the **SET_LOOP_TASK_STACK_SIZE(10240);**  line and the software compiles without error.   

Eventually worked out that you need Arduino IDE 2 .0 or greater, and then update all the libraries so all the underlying code is there, posting here to ask Joseph to add it to the docs, and for others to see and know the fix  if they see an issue

gjchester, 2023-08-29:
Closing as this was a FYI for others.

closed 2023-08-29 by gjchester

JosephHewitt, 2023-08-30:
I'm not sure what was causing your problem but it compiles okay for me using Arduino IDE 1.8.19, I've never needed to use Arduino IDE 2. Commenting out that line will prevent OTA updates from working on your device unless you disable the checksum verification as documented in #86.

gjchester, 2023-08-30:
Pass, All I know was updating to a new Arduino IDE fixed it for me...