JosephHewitt/wardriver_rev3 #118: Temp not displaying on 1.2.0b4

d0p4m1n31, 2023-10-01:
Running the 1.2.0b4 version beta software, but cannot get the temp to show on the display. Is this a known issue? Like in 1.0.3? To make sure it is not hardware, I replaced the temp sensor and resistor

JosephHewitt, 2023-10-01:
Hi,

No this isn't a known issue. I have attempted to reproduce this on 2 of my test devices and they both work as expected on 1.2.0b4. Keep in mind it may take a few moments to appear after you turn on the device but it should be there within a minute.

Could you connect your PC to the ESP "side B" and look at the serial monitor on baud 115200? When it first turns on, you should see some text scrolling by which will log the status of the temperature sensor (along with other things). The message `Unable to detect DS18B20 temperature sensor!` indicates the sensor cannot be communicated with, and `DS18B20 detected with ID = [...]` indicates it was detected.

The temperature is then sent to "side A" to be displayed on the LCD. By checking this we can determine where the problem is occuring which might narrow down the problem. 

Are you also willing to share an image of how you soldered the temperature sensor and resistor? I can then check the placement etc.

JosephHewitt, 2023-10-19:
Had a conversation regarding this outside of GitHub and this issue is likely hardware related.

closed 2023-10-19 by JosephHewitt