JosephHewitt/wardriver_rev3 #122: Idea: Total wifi captured on LCD

symph0nic, 2023-10-23:
What it says on the subject - would be cool to see the 'status' of the current session by showing the total number of wifi networks captured on the LCD, rather than (or as well as) the number currently in range. 

Just an enhancement idea...thanks!

JosephHewitt, 2023-10-23:
Hey,

Thanks for the suggestion, I like the idea. The only caveat is that the wardriver doesn't keep a full list of discovered networks while running (it only keeps the last 512). This means it might start counting some networks more than once if you travel back into an area which you have already been in.

I don't think that is a giant problem though and will work on this in the future.

JosephHewitt, 2024-10-05:
This is now implemented, and will be included in the next release (likely 1.2.1).

Thanks for the suggestion, and sorry it took so long to implement.

PR #184 (2024-10-05): Add total seen WiFi count
Closes #122

Adds a value "T" (short for total) on the LCD line 3 which shows the count of unique WiFi networks discovered.
 
Note: The wardriver only knows if a network is new if it was not one of the previous 512 networks seen. This means that the number here could be higher than as reported by WiGLE, especially if the wardriver returns to an area previously visited after capturing 512+ networks elsewhere.
closed 2024-10-05 by JosephHewitt