JosephHewitt/wardriver_rev3 #144: Text doesn't seem aligned on the OLED screen

hyuma78, 2024-01-26:
Hi, I just built my own wardriver, but I have some strange issue. When the texts appears on my oled 0.91 128x32px it seems like not aligned as you can see on the screenshot that I have attached here. I resoldered, change wires, test continuity between pads of the OLED but no luck. I also repogrammed 2-3 times the ESP32 but same result. I also tested the I2C protocol of the screen with an arduino sketch and it's "3c". I tried to replace the screen but I have the same issue. At this point I need your help to try to troubleshooting this. Thanks in advance!

[image]
[image]
[image]

hyuma78, 2024-01-26:
No, I don't have any other ESP32, except for that problem, the rest seems to work, do you think it depends on some library that I installed? I'm using Arduino IDE 2.2.1, and I didn't tried by compiling myself yet.

JosephHewitt, 2024-01-28:
If you install from binary, it will rule out any problems caused by libraries etc. 

My gut feeling is this is either a faulty ESP32 or a bad batch of LCDs.

hyuma78, 2024-01-31:
I've done a really insane thing :D ... I basically hot air desoldered every components of the esp32 (resistors, ic, etc...) and resoldered all together and now it seems that there are no problems, I also edited the code to works with 128x64 display. I will keep you updated if this issue is resolved

JosephHewitt, 2024-04-10:
Hi,

Thanks for informing us on how you fixed the issue. Please feel free to reopen this issue if the problem comes back, but I am assuming this is fixed now based on your last comment.

closed 2024-04-10 by JosephHewitt