JosephHewitt/wardriver_rev3 #152: After factory reset, Wigle auto-upload fails to upload

pejacoby, 2024-05-09:
I did a factory reset which set my boot count (bc) value back to zero.  I updated to 1.2.0rc1 in the process. Now, despite having Wigle.net automatic uploads enabled, the system checks Wigle and doesn't upload anything.

My filenames started over at "wd3-6.csv", down from "wd3-543.csv"

Can I reset the boot count value so files will not be recognized by Wigle as already uploaded?  Maybe just leave a file on the SD card named "wd3-544.csv" ?

From test.txt:
_BOOT_1.2.0b5, ut=2345679, rr=1, id=3893376, bc=543, ep=1714850993, bsh=xxxxx
_BOOT_1.2.0b5, ut=2406328, rr=1, id=3893376
_BOOT_1.2.0b5, ut=2381336, rr=1, id=3893376
_BOOT_1.2.0rc1, ut=2352970, rr=3, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2423558, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2357448, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2345806, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2292952, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2331731, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2332715, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2338596, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2428653, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2484996, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2419580, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2408088, rr=1, id=3893376, bid=0, bc=6, ep=1714932725, bsh=xxxxx
_BOOT_1.2.0rc1, ut=2419047, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2388787, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2384958, rr=1, id=3893376, bid=0
_BOOT_1.2.0rc1, ut=2385615, rr=1, id=3893376, bid=0, bc=10, ep=1715032422, bsh=xxxxx
_BOOT_1.2.0rc1, ut=2439587, rr=1, id=3893376, bid=0, bc=11, ep=1715050381, bsh=xxxxx
_BOOT_1.2.0rc1, ut=2435770, rr=1, id=3893376, bid=0, bc=12, ep=1715204319, bsh=xxxxx
_BOOT_1.2.0rc1, ut=2439979, rr=1, id=3893376, bid=0

JosephHewitt, 2024-05-12:
Hi,

Thanks for the report. I had a look at potential causes for this and have found a bug which is likely the cause of your issue.

When the auto upload feature is enabled, it saves the current `bc` value separately and will refuse to auto upload any files with an ID lower than this. It is implemented this way as a safety net to prevent a large number of files being uploaded to WiGLE in the event that it didn't correctly detect that the old files were already uploaded. 

The reason why I think this went wrong for you is because during a factory reset, the old `bc` value is still kept in memory until the next reboot. So if you reset the device, then immediately set it up again, it would have used the old value as the safety net (`543` as you mention in your post), so no files with an ID lower than this can be uploaded.

You can fix this on your wardriver by performing another factory reset and then rebooting it before you configure it again.

This isn't intentional behavior and I will fix this ready for the next release.

As for the `bc` value, it is not possible to increase it by custom amounts. I have realized there are some issues with the current method of resetting the `bc` value and have opened a separate issue for that: #154

pejacoby, 2024-05-12:
Thanks! I'm here to find the edge cases :)

PR #160 (2024-06-30): Full restart on factory reset, increment bootcount to match filesystem
Closes #152 by performing `ESP.restart()` during factory reset
Closes #154 by incrementing `bootcount` when a colliding filename is detected
commit 6f73271b (2024-06-30): Full restart on factory reset, increment bootcount to match filesystem (#160)

Closes #152 by performing ESP.restart() during factory reset
Closes #154 by incrementing bootcount when a colliding filename is detected
closed 2024-06-30 by JosephHewitt