JosephHewitt/wardriver_rev3 #156: SSID character safety - some CSV breaking examples of embedded , and ^M

pejacoby, 2024-05-19:
I found a few examples of BLE and WIFI naming with characters that break the CSV file by forcing an extra comma or end-of-line. Curiously a number seemd tied to the same kind of device.

Extra comma before the last character:

> 6F:62:D6:E9:C4:5F,WiZDOéâ,é,[BLE],2024-04-26 22:47:53,0,-100,41.7441406,-72.6834946,26.30,2.50,BLE
> 6C:57:61:C0:A9:F7,WiZDOéâ,é,[BLE],2024-04-27 03:11:19,0,-94,41.7441673,-72.6834869,32.80,2.75,BLE
> 73:03:E1:9B:47:8C,WiZDOéâ,é,[BLE],2024-04-27 03:20:59,0,-99,41.7441597,-72.6834793,34.40,2.75,BLE
> 73:03:E1:9B:47:8C,WiZDOéâ,é,[BLE],2024-04-27 03:30:49,0,-97,41.7441635,-72.6834946,34.80,2.75,BLE
> 

None of the ID's lookup but it appears this are mobile and might be this - https://www.amazon.com/ZIIDOO-Bluetooth-Transmitter-Receiver-Wireless/dp/B0829S6K1X
They present with all kinds of crazy characters after the "WiZDO..." but only the ones with the caret-a/comma/accent-e are breaking lines.

Examples:
> WiZDOéè\D
> WiZDOéÜÀö
> WiZDOéè”
> WiZDOée˛
> WiZDOééÎ6
> WiZDOéè‘
> WiZDOéÑØò
> WiZDOééÌ
> WiZDOéèf†

Embedded end-of-line in the name.
 The first three appear to be a Z-Link backup camera or similar. The last one is a one off EspressIf board of some sort that I spotted.

> 94:A3:CA:0C:0E:42,Z-Link_0E42^M,[BLE],2024-04-22 12:31:59,0,-90,42.3310242,-83.0412064,51.70,9.25,BLE
> 60:73:BC:E0:AE:20,Z-Link_AE20^M,[BLE],2024-04-22 22:27:34,0,-95,42.3321686,-83.0413055,186.70,3.00,BLE
> D0:5B:A8:58:5E:0F,Z-Link_5E0F^M,[BLE],2024-04-23 02:36:47,0,-96,42.3318863,-83.0411911,94.80,4.25,BLE
> 44:17:93:DC:C3:11,FSBT-441793DCC312^M,[WPA_WPA2_PSK],2024-04-23 12:28:16,1,-87,42.9476967,-78.6766510,238.50,3.00,WIFI

pejacoby, 2024-06-09:
For the extra-comma BLE examples, I was looking at parse_bside_line() and it seems a fix could be as simple as adding this just before the value is put into the 'out' string - the same approach as in the WiFi scan loop.

    ble_name.replace(",","_");

It could also go at line 2807 just after the ble_name is grabbed from the buffer, but then users would need to know to set any BLE BLOCK entries up with underscore character replacing a comma in the name.

For the "^M" cases, would a similar replace work? I'm not totally familiar with what the .replace method can touch.

ble_name.replace("\r","_");

JosephHewitt, 2024-10-05:
Thanks for the report. I have added a replacement for commas so they will be escaped in the same way as SSIDs currently are.

I want to implement a more versatile way of escaping strings (instead of lots of `.replace` calls) so that lots of problematic characters can be replaced at once. I was thinking a function which takes the numerical ordinal of each char, and replaces everything between `x` and `y` with an underscore. I'll investigate this more in the future, however.