JosephHewitt/wardriver_rev3 #157: Trying to flash board B and getting error while compiling sketch

ihardng, 2024-05-31:
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), DIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

Downloads\wardriver_rev3-1.1.0\B\B.ino: In function 'void loop()':

B:239:48: error: conversion from 'BLEScanResults*' to non-scalar type 'BLEScanResults' requested

  239 |   BLEScanResults foundDevices = pBLEScan->start(2.5, false);

      |                                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~

Multiple libraries were found for "WiFi.h"

 Used:AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\WiFi

 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

conversion from 'BLEScanResults*' to non-scalar type 'BLEScanResults' requested

board A flashed just fine.

SassyNarwhal, 2024-06-01:
I'm having the same issue.

JosephHewitt, 2024-06-01:
Please can you check if you have the correct ESP32 board installed? Version 2.0.12 should be used.

SassyNarwhal, 2024-06-01:
That seemed to work for me. I didn't realize that the older version needed to be used. Thanks for the help!

ihardng, 2024-06-02:
> Please can you check if you have the correct ESP32 board installed? Version 2.0.12 should be used.

worked for me as well. Thank you!!

closed 2024-06-02 by ihardng