JosephHewitt/wardriver_rev3 #158: Not connecting to WiFi or creating backup network

bigblzrd, 2024-06-16:
I've reflashed a few times and everything else functions perfectly but I can't seem to connect to my network after the first time setup and it doesnt create the backup network.

I also tried looking at the serial output upon boot but nothing really stood out to me -
```
Waiting for config vars
IN:
IN:PUSH:sb_bw16=

IN:
IN:
IN:
IN:
IN:
DS18B20 detected with ID = 28 61 64 34 8F 57 C4 2D
Requesting temperature
Requesting data from SIM
AT+CNETSCAN=1

OK

Setting up Bluetooth scanning
Setting up multithreading
Started
```
However after this sequence it goes straight in to scanning while the screens prompting me to connect to the backup network. Any help with this would be greatly appreciated!

JosephHewitt, 2024-06-16:
Hi,

What exactly does the LCD show during this?

Could you also check the serial output of the other ESP (ESP A)? That is the one responsible for hosting/connecting to the AP.

Thanks

bigblzrd, 2024-06-16:
> Hi,
> 
> What exactly does the LCD show during this?
> 
> Could you also check the serial output of the other ESP (ESP A)? That is the one responsible for hosting/connecting to the AP.
> 
> Thanks

Hi there, 

Oversight on my part, I didn't even think about the output of the other board haha but here it is - 

```
Starting v1.2.0rc1
Chip ID: 16055360
SD Card Type: SDHC
SD Card Size: 7618MB
Board ID = 0
Sending config options to B..
cfg.txt could not be opened
Pushing config sb_bw16
Setting/loading boot config..
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
Loaded variables
Board ID = 0
Device already identified as rev3
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
cfg.txt could not be opened
Attempting to connect to WiFi...................................................
Disconnecting
```
Seems I'm missing a config file, during this sequence the LCD just prompts me to connect to the backup network but it's unable to create it which now makes sense, I must've missed something in the setup process?

JosephHewitt, 2024-06-16:
I don't see anything unusual in the output above aside from failing to connect to your WiFi which could be caused by a lot of different things (bad credentials, too weak of a signal, etc). That config file is optional so it doesn't matter if it can't be opened. 

You say it is unable to create the backup network, but the output above doesn't say that. Are you sure the network doesn't get created? You should be able to connect to it with a phone or laptop. Did you configure a unique name for the backup network? If the name is the same as another nearby network that could also cause issues.

How did you flash the device? If you compiled it yourself, be sure that you have the ESP32 board version 2.0.12 installed since later versions have introduced some issues which could potentially cause WiFi issues.

bigblzrd, 2024-06-16:
> I don't see anything unusual in the output above aside from failing to connect to your WiFi which could be caused by a lot of different things (bad credentials, too weak of a signal, etc). That [...]

Hmm, I double checked the credentials I entered for my home network multiple times and I made sure to give the backup network a unique name and password. It just fails to connect and then when the screen prompts the backup network it doesn't appear anywhere on my phone or desktop.

I used the Arduino IDE to flash and I believe I used the correct versions for everything, I'll have to give it another go and see if I went wrong anywhere and report back!

JosephHewitt, 2024-06-16:
One more thing, it is possible that the signal strength of the ESP-A is very weak. I've seen this happen before with defective units, but it can also be caused by not plugging in the antenna fully (or not connecting one at all).

bigblzrd, 2024-06-16:
> One more thing, it is possible that the signal strength of the ESP-A is very weak. I've seen this happen before with defective units, but it can also be caused by not plugging in the antenna fully (or not connecting one at all).

I've tried with and without antennas to no avail, is it possible this could be caused by me using the USB-C variant of the ESP32 boards at all? Doesn't seem like that should be a problem but I'm just thinking of anything at this point haha

bigblzrd, 2024-06-18:
So I found a temporary solution.

If I use a config file with the credentials it works flawlessly, I'll have to do some more digging at some point to find the true cause of the issue!

JosephHewitt, 2024-10-05:
Hi,

I did some investigation into the WiFi setup today and found something which could cause the wardriver to fail to connect to a network when configured normally, but work as expected when using the config file. This bug would be triggered if your WiFi password contains a `%` character.

However, I'm still unsure why the fallback network was not showing up for you.

I've merged the fix into `main` with #187 and it will be included in the next software release (likely 1.2.1).

bigblzrd, 2024-10-05:
> Hi,
>
> I did some investigation into the WiFi setup today and found something which could cause the wardriver to fail to connect to a network when configured normally, but work as expected when using [...]

Hey! Nice to hear an update on this issue, your findings with the bug being triggered by the  ```%``` character lines up perfectly in my case as my password did contain one!