JosephHewitt/wardriver_rev3 #164: SD card not being recognized

ihardng, 2024-07-29:
I have the war driver assembled according to the directions
boards A and B have been flashed, still getting an error that there is no SD card
checked all soldering joints, installed new SD card reader, reflashed boards with both new firmware and previous firmware, tried multiple 16GB SD cards, and formatted both cards multiple times.

Does anyone have any other troubleshooting tips that might be worth a try?

JosephHewitt, 2024-07-31:
Are you formatting the SD cards with FAT32? You cannot use exFat or any other filesystems.

Have you tried another brand of SD card? In some situations, particular brands/models of cards aren't compatible but it's hard to make an exhaustive list of what does/doesn't work.

Do you definitely have the same SD card reader module as listed on the website? Some of the other modules have different operating voltages and won't work.

Have you looked at the serial output of ESP "A" on boot? You can see this with the serial monitor on the Arduino IDE at baud 115200. This will show whether the SD card was detected or not, which could help track down the issue.

Finally, have you tried flashing the ESPs from a binary release? (eg, using esptool rather than the Arduino IDE). This can rule out issues potentially caused by having wrong libraries installed on your PC while compiling yourself.

If none of the above fix the issue, it could potentially be a bad ESP32 module.

ihardng, 2024-08-03:
Hi Joseph,
Thanks for getting back to me. To answer your questions...

Yes I have 2 different brands of SD cards. Both are 16gb, formatted in FAT32.

The SD card reader is the same module as listed on the website.

I will give the other steps a try and let you know the outcome.

ihardng, 2024-08-11:
I reflashed within in Arduino, and monitored board A from the serial monitor, below is all I get

SD Begin failed!
No SD card attached!
SD Card Type: UNKNOWN
SD Card Size: 0MB
Failed to open file for writing.
Failed to open file for writing.
Failed to open file for writing.
Failed to open file for writing.
Failed to open file for writing.
Failed to open file for writing.
Failed to open file for writing.
Failed to open file for writing.

I tried to flash from binary, but unfortunately that is over my head. I got python installed but couldn't install the esptool

JosephHewitt, 2024-08-11:
Hi,

Thanks for the information. Since the card isn't detected at all, it's less likely to be related to anything "high level" like the filesystem or software. For that reason, I suspect the likely cause is either a faulty ESP32, faulty card reader, or a bad solder joint somewhere. Unfortunately these are the hardest types of issues for me to help with remotely.

You said you checked all solder joints, but did you also check the ones on the ESP boards themselves? I have seen some of them arrive badly soldered from the factory.

> just as a test, I connected board B and opened the serial monitor and I can see it identifying networks and devices next to me. There was a lot of activity on the serial monitor when connected to B.

ESP-B will happily send over all the networks it finds even if there is a fault on ESP-A.

ihardng, 2024-08-12:
Hi Joseph, 
It seems on of the joints on the esp was the cause. I resoldered all of them very carefully and everything is now functioning as it should. Thank you for the reminder and quick response!