JosephHewitt/wardriver_rev3 #165: Compilation error on board B

PythonEatsSQuirreL, 2024-08-04:
I am getting error when trying to upload the B board on BLEScanResults foundDevices = pBLEScan->start(2.2, false);

Error:

Compilation error: conversion from 'BLEScanResults*' to non-scalar type 'BLEScanResults' requested

Any ideas?

PythonEatsSQuirreL, 2024-08-04:
it throws the error on verifying also so I guess it's not board related

JosephHewitt, 2024-08-04:
Hi,

The ESP32 board library was recently updated with breaking changes that are not yet supported by this project, only the 2.x branch is supported.

The suggested version is 2.0.12 which you can install from `Tools -> Board -> Boards Manager`.

The 3.x branch will be supported in a future release.

PythonEatsSQuirreL, 2024-08-04:
Thank you for the incredible fast solution!

It works with the version specified by you.

Cheers mate!

closed 2024-08-04 by PythonEatsSQuirreL