JosephHewitt/wardriver_rev3 #179: 1.2.1 Only Picks Up APs When connected to my Wifi

RicklePick3000, 2024-09-21:
Just built a JH wardriver, and all is functional, set up and as should be. When connected to my Wifi, the device picks up AP in the area i know to exist, but as soon as the device is away from a connection to my wifi, it no longer picks up AP only BLE and GPS sats.

JosephHewitt, 2024-09-21:
Thanks for the report. This could possibly be related to #178. I am attempting to reproduce these issues to find the cause.

Please keep in mind that 1.2.1 is not an official release, but is the development version containing the latest untested changes. I suggest running 1.2.0 unless you specifically want to test these latest changes.

JosephHewitt, 2024-09-21:
I was able to reproduce this and it appears to have the same root cause as #178.

I have merged a fix in #180.

Thanks again for the report. Please feel free to reopen this issue or create a new issue if you see a recurrence of this.

closed 2024-09-21 by JosephHewitt