JosephHewitt/wardriver_rev3 #200: SD Begin failed - Maybe PCB?

pvtcompyle, 2025-08-05:
I connected the SD module to an external esp32 to verify it worked and had no issues. Then I directly connected esp32-A to the SD card module with jumpers and it worked fine. Removing one jumper at time, I was able to deduce that there's something going on with the CS and CLK connections (pins 18 and 5) to the SD module. I've checked the solder joints and even swapped the esp32-A board with a different one, but no dice.

I'm at point of thinking maybe it's the PCB, but I was also thinking maybe the CS and CLK circuits are routing through another chip on their way. I've checked the other solder joints and they all seem fine. I don't suppose anyone has any insights regarding the CS and CLK passing through other components?

JosephHewitt, 2026-01-17:
Hi,

Sorry for the delay in looking into this.

> Removing one jumper at time, I was able to deduce that there's something going on with the CS and CLK connections (pins 18 and 5) to the SD module

Could you elaborate more on what you actually did/observed here? Removing any of the jumpers should cause an `SD begin failed` situation since all of the connections are required and in use. What leads you to believe the issue is specifically related to `CS` or `CLK`?

As for the PCB routing, I think it's unlikely to be the cause since I haven't yet seen any faulty boards (mostly due to how simple the design is). Since it is a 2-layer PCB, if you take a good quality photo of both sides of the board, then all of the traces should be visible and any manufacturing failures serious enough to cause this should be visible.

`SD Begin failed` can also be caused by the card itself, some can be very inconsistent. Did you try any other brands? In my experience, the cheaper brands and smaller capacities often work better.

Did you also try replacing the SD module? Some of them don't always grip the SD card well enough to make a strong connection to the card, meaning that vibrations or moving the unit can cause it to disconnect in a way that isn't easy to see.

pejacoby, 2026-02-05:
For what it's worth, one of my SignalSleuth builds has gone into SD Begin Failed mode after a lot of use, it definitely feels like the module isn't holding the card tightly anymore.  I need to unsolder and replace with a new one at some point.

pvtcompyle, 2026-02-05:
While  both MCU's were soldered to the PCB with all he components, I directly jumped from the MCU pins to the SD card reader. When  I did that, it was able to read the SD normally. This effectively bypassed the PCB entirely. Then I removed one jumper at a time allowing the connection to go through the PCB circuitry. Removing the CS or CLK jumpers were the only ones that caused the SD card to stop working.  

I did replace the SD module, but it made no difference and the original module worked when plugged into a different environment. I also swapped different size/brand SD cards. Ultimately though, since the module and card worked when the PCB was bypassed, I'm pretty confident that isn't the issue. 

Oddly enough though, in the process of writing this, I decided to upload the sketches again. For reasons I cannot explain, the SD card issue seems to be gone. I got it once then rebooted and it has been fine for several hours with multiple reboots. The only thing I did differently is I recently switched from Windows to Ubuntu.  I'm not sure what to make of that.